SignalK/SensESP #508: Using SensESP as PlatformIO library related issues

pstolarz, 2021-12-31:
While working on [PR](https://github.com/SensESP/OneWire/pull/3) for [SensESP/OneWire](https://github.com/SensESP/OneWire) lib, which uses SensESP as external library I encountered the following compilation and linking errors:
* Initialization reorder warning treated as an error. For example here:
```
In file included from .pio/libdeps/esp32dev/SensESP/src/sensesp/net/networking.cpp:1:0:
.pio/libdeps/esp32dev/SensESP/src/sensesp/net/networking.h: In constructor 'sensesp::Networking::Networking(String, String, String, String, const char*)':
.pio/libdeps/esp32dev/SensESP/src/sensesp/net/networking.h:64:15: error: 'sensesp::Networking::wifi_manager_password_' will be initialized after [-Werror=reorder]
   const char* wifi_manager_password_;
               ^
.pio/libdeps/esp32dev/SensESP/src/sensesp/net/networking.cpp:26:19: error:   base 'sensesp::Startable' [-Werror=reorder]
       Resettable(0) {
                   ^
.pio/libdeps/esp32dev/SensESP/src/sensesp/net/networking.cpp:21:1: error:   when initialized here [-Werror=reorder]
 Networking::Networking(String config_path, String ssid, String password,
```
* Improper ESP32 partitions configuration during linking time:
```
Flash: [=======Error: The program size (1548656 bytes) is greater than maximum allowed (1310720 bytes)
=*** [checkprogsize] Explicit exit, status 1
==]  118.2% (used 1548656 bytes from 1310720 bytes)
```

**Root cause**
While using SensESP as standalone app project both of these errors are mitigated by the following `platformio.ini` configuration:
```
[espressif32_base]
;this section has config items common to all ESP32 boards
platform = espressif32
build_unflags = -Werror=reorder
board_build.partitions = min_spiffs.csv
```
Unfortunately `platformio.ini` is not considered while using SensESP as external library.

**Fix proposition**
* Reorder compilation errors may be fixed by initialization lists reordering in some constructors (I may provide PR for this if you would like) OR add the following to the `library.json`:
```
"build": {
  "unflags": "-Werror=reorder"
}
```
* Partitions configuration needs to be added directly to the final app project's `platformio.ini` exactly the same way as in SensESP's counterpart of this file. In this case SensESP documentation should be updated to point this out for future SensESP v2 users.

Boatingwiththebaileys, 2022-01-16:
I've just started to get these issues when I compile, and I didn't before.  I am seeing a lot of this:
.pio/libdeps/firebeetle32/SensESP/src/signalk/signalk_metadata.h:25:10: error: 'SKMetadata::units_' will be initialized after [-Werror=reorder]

Where do I add that line, tin the SensESP/library.jason file or the route project? Also is it just pasted in at the end?

pstolarz, 2022-01-17:
@Boatingwiththebaileys  The described problem pop-ups on developed SensESP v2 (`v2-dev` branch). It shouldn't take place on v1. I'm not sure what version you are talking about. The `-Werror=reorder` needs to be added to SensESP `library.json` file, e.g. like here:
```diff
diff --git a/library.json b/library.json
index f0f4bfc8..81f92029 100644
--- a/library.json
+++ b/library.json
@@ -59,6 +59,9 @@
       "version": "https://github.com/JoaoLopesF/RemoteDebug.git#0b5a9c1a49fd2ade0e3cadc3a3707781e819359a"
     }
   ],
+  "build": {
+      "unflags": "-Werror=reorder"
+  },
   "version": "2.0.0-alpha",
   "frameworks": "arduino",
   "platforms": "*"
```

closed 2023-11-20 by pstolarz