SignalK/SensESP #509: Noisy switches and DigitalInputCounter

smr547, 2022-01-05:
I'm wiring up a DigitalInputCounter to a rain gauge so that I might count the number of pulses from the gauge. Unfortunately, the rain gauge reed switch is noisy, triggering multiple interrupts per event. It takes a few microseconds for the switch to settle down again. I'd like a counter that ignores events that occur within some time interval of a previously recorded event. Should DigitalInputCounter have an optional parameter that implements this behavior?

I spent some time trying to get this work with the debounce filter, but I don't believe this implements the behavior I want.

smr547, 2022-01-05:
This worked nicely for my use case:

```patch
diff --git a/src/sensors/digital_input.cpp b/src/sensors/digital_input.cpp
index 28ea41c8..75f69b52 100644
--- a/src/sensors/digital_input.cpp
+++ b/src/sensors/digital_input.cpp
@@ -53,14 +53,27 @@ bool DigitalInputState::set_configuration(const JsonObject& config) {
 DigitalInputCounter::DigitalInputCounter(uint8_t pin, int pin_mode,
                                          int interrupt_type, uint read_delay,
                                          String config_path)
+    : DigitalInputCounter(pin, pin_mode, interrupt_type, read_delay, 0, config_path) {}
+
+DigitalInputCounter::DigitalInputCounter(uint8_t pin, int pin_mode,
+                                         int interrupt_type, uint read_delay,
+                                         uint ignore_period_ms,
+                                         String config_path)
     : DigitalInput{pin, pin_mode, interrupt_type, config_path},
       IntegerProducer(),
-      read_delay_{read_delay} {
+      read_delay_{read_delay},
+      ignore_period_ms_{ignore_period_ms}, last_event_time_ms_{0} {
   load_configuration();
 }
 
 void DigitalInputCounter::enable() {
-  app.onInterrupt(pin_, interrupt_type_, [this]() { this->counter_++; });
+  app.onInterrupt(pin_, interrupt_type_, [this]() {
+    unsigned long now = 0;
+    if (ignore_period_ms_ == 0 || ((now = millis()) - last_event_time_ms_) > ignore_period_ms_) {
+      this->counter_++;
+      last_event_time_ms_ = now;
+    }
+  });
 
   app.onRepeat(read_delay_, [this]() {
     noInterrupts();
diff --git a/src/sensors/digital_input.h b/src/sensors/digital_input.h
index 9684f1cb..15eab7af 100644
--- a/src/sensors/digital_input.h
+++ b/src/sensors/digital_input.h
@@ -84,6 +84,9 @@ typedef DigitalInputState DigitalInputValue;
  * 
  * @param read_delay How often you want to read the pin, in ms
  * 
+ * @param ignore_period_ms Ignore events occuring within this interval after
+ * a recorded event.
+ *
  * @param config_path The path to configuring read_delay in the Config UI
  * 
  * */
@@ -92,10 +95,15 @@ class DigitalInputCounter : public DigitalInput, public IntegerProducer {
   DigitalInputCounter(uint8_t pin, int pin_mode, int interrupt_type,
                       uint read_delay, String config_path = "");
 
+  DigitalInputCounter(uint8_t pin, int pin_mode, int interrupt_type,
+                      uint read_delay, uint ignore_period_ms, String config_path = "");
+
   void enable() override final;
 
  private:
   uint read_delay_;
+  uint ignore_period_ms_;
+  unsigned long last_event_time_ms_;
   volatile uint counter_ = 0;
   virtual void get_configuration(JsonObject& doc) override;
   virtual bool set_configuration(const JsonObject& config) override;
```

mairas, 2022-01-06:
Hi! I was going to suggest you to check out the Debounce transform, but after reading your description again, I realized you want to count the pulses instead of just getting the new state. You're right that in that case the functionality would probably have to be included in the sensor.

I wouldn't want to add any complexity to DigitalInputCounter interrupt function. It is used to measure frequencies of many signals that may be in the range of several kHz, and any added conditional evaluation will lower the maximum frequency the sensor can reliably measure.

Instead, you should be able to inherit the DigitalInputCounter class and add the changed functionality in the derived class. This should be quite simple to do. Also, one further consideration is that you might want to have a look at the `v2-dev` branch. We've been preparing a new major version release and there are some changes to DigitalInputCounter as well. If you want to be as future-proof as possible, you might want to implement the new features against that branch.

smr547, 2022-01-07:
Thanks for the feedback. I've created https://github.com/SignalK/SensESP/pull/510 to address this. I back-ported this counter to the v1 API as well. I haven't ported my application to v2-dev branch, so I haven't been able to test this patch at all (it might not even compile).

mairas, 2022-01-31:
Fixed in PR #510.

closed 2022-01-31 by mairas