SignalK/SensESP #512: How to specify SDA, SCL for I2C?

nadrog, 2022-01-08:
Hi, can you please explain what's the right way to define specific SDA and SCL pins for I2C? Thanks!

ba58smith, 2022-01-08:
Version 1 of SensESP's Sensors doesn't work with non-standard I2C pins. Version 2.0 is going to completely change how SensESP works with I2C sensors, but version 2.0 is still a WIP.

If you tell me the specific I2C sensor you're working with, I'll try to give you the code to make it work with version 1.

nadrog, 2022-01-08:
Actually, I found out that if I simply configure Wire before sensor is
instantiated, it all works. My code now looks like following:
  Wire.begin(SDA_PIN, SCL_PIN, FREQUENCY);
  auto* bme280 = new BME280(0x76);

ba58smith, 2022-01-08:
Excellent. I'll close this.

closed 2022-01-08 by ba58smith