SignalK/SensESP #523: RFE: Add support for Zigbee and Z Wave

mgrouch, 2022-01-18:
Add support for Zigbee  and Z Wave

Zigbee and Z Wave looks like more suitable protocol for a boat sensors than WiFi.

Thanks

JohnySeven, 2022-01-19:
Why would you do that? Future of SensESP is mainly around ESP32 and is designed as library for connecting to SignalK. Why wasting WiFi for Zigbee or Z Wave that are less capable? Better would be implementing SignalK via CAN - this is something @mairas has in his long term plan. Or maybe implement communication via BLE would be also interesting implementation as it's supported by hardware on ESP32 side and RPI side.

Would be good if you provide more reasons why in your post. What you want to achieve with implementation? What would be benefits of that implementation to whole?

Thanks

dmtamsen, 2022-01-19:
Another thing to take into account could be MQTT - then you can have the integration between Homeassistant and SignalK server.

EdKok, 2022-01-19:
> Why would you do that? Future of SensESP is mainly around ESP32 and is designed as library for connecting to SignalK. Why wasting WiFi for Zigbee or Z Wave that are less capable? Better would be [...]

Zigbee and Z-Wave do not make use of WiFi itself, but the Zigbee protocol operates at the same 2.4 GHz band while Z-Wave operates on the low-frequency 908.42 band. So I think Zigbee is indeed not a good idea and Z-Wave is expensive.
BLE is also not without its own problems. It doesn't reach very far and enabling it on a RPi is known to interfere with its own WiFi on 2.4 GHz. So also a not a good idea. But good a SensESP module (ESP32) act as a gateway between BLE and WiFi maybe? So that it communicates over BLE with devices such as ruuvi-tags and sends the data via WiFi to SignalK.

mairas, 2022-01-19:
Not in focus. Closing.

closed 2022-01-19 by mairas

mairas, 2022-01-19:
> But could a SensESP module (ESP32) act as a gateway between BLE and WiFi maybe? So that it communicates over BLE with devices such as ruuvi-tags and sends the data via WiFi to SignalK.

I think this would be feasible. But the implementation can and should be a separate project.