SignalK/SensESP #524: RFE: Add integration with homeassistant

mgrouch, 2022-01-19:
Add integration with homeassistant.

Homeassistant being more generic is more supported than SignalK and has way bigger community.

Thanks

dmtamsen, 2022-01-19:
Will esphome.io not better suit your needs then?

mgrouch, 2022-01-19:
Thanks. I didn’t know it existed. 
Looks awesome by reading. 
Homeassistant automations capabilities are much better than SignalK. 
Now a question why SensESP team is not partnering with esphome.io?
I think it be better than some of 
SignalK components and some of SensESP be integrated into homeassistant and esphome.io. 

Again boat is much like a home with some more specific needs. Home automation community grows rapidly and it’s much bigger than boating. 
I think boating software should just integrate more into it to avoid reinventing stuff which will be tested less overall (less users)
Just random thoughts. 

thanks for the link, I’ll study more

mairas, 2022-01-19:
Not in focus. Closing.

closed 2022-01-19 by mairas