SignalK/SensESP #552: Authorizing with SignalK

nadrog, 2022-02-11:
Hi,

  with fresh SignalK server installation and SensESP 2.1.1, my SailorHat is trying to authenticate with SignalK server (that was succesfully found). SK connection status on SensESP is "Authorizing with SignalK" and debug message is "(C1) Attempting to connect to Signal K Websocket...", but I do not know what I need to do to authenticate this new device with SignalK server. Please, point me to appropriate documentation. Thanks!

Gordan

nadrog, 2022-02-11:
Further info: "Allow New Device Registration" is enabled in SignalK and there is no inputs in "Access Requests"

nadrog, 2022-02-11:
I used the approach defined in an old version of Troubleshooting page found on Way Back Machine (https://web.archive.org/web/20211217183358/https://github.com/SignalK/SensESP/wiki/Troubleshooting). Can you please, add this info back? Thanks!

mairas, 2022-02-11:
I was just going to comment that resetting the device might help. But nevertheless, the same troubleshooting information still exists: https://signalk.org/SensESP/pages/troubleshooting/. Instead of writing elaborate workaround instructions for bugs that I myself can't reproduce, I'd prefer getting as detailed description of the bug and the environment so that the bug could be reproduced or at least narrowed down.

For that, could I get:
- Copy of the serial monitor output
- Server version
- Server configuration
- Related server log entries, if any

nadrog, 2022-02-13:
> I have been able to reproduce similar behavior by removing the device access on the server side. The device doesn't recognize that the access has been removed and keeps on reconnecting with the old token.

Hi Matti,

 

  Yes, that’s how I got into this problem: I reinstalled the SignalK.

 

Br, G

nadrog, 2022-02-13:
My problem was finding the troubleshooting section when I started from SensESP Git page. This could potentially be improved.

mairas, 2022-02-13:
> My problem was finding the troubleshooting section when I started from SensESP Git page. This could potentially be improved.

Maybe you could give more concrete suggestions how to improve on that? I think the documentation site is quite prominently linked to in the README, and also in the "About" sidebar text.

The wiki is deprecated and only used for the legacy v1 documentation. I just added a link to the documentation site on the wiki home page, though.

nadrog, 2022-02-13:
I believe the link you just added is the only missing thing.

mairas, 2022-02-13:
Accidentally closed; the actual reported issue remains unfixed.

PR #559 (2022-02-15): Fix SK access token validity checking
The previous SK server endpoint for testing the SK access token validity didn't work, resulting in the device no longer being able to connect to the server if the access token was invalidated on the server.

The validity checking endpoint at the server is not implemented, but I found out that the websocket stream endpoint works for this purpose, at least somewhat.

Some related SK server issues:

- SignalK/signalk-server#1396
- SignalK/signalk-server#1397

Fixes #552.
closed 2022-02-15 by mairas