SignalK/SensESP #587: Wifi credentials cannot be changed

qMalte, 2022-04-14:
While programming the ESP32 with the SensESP library I encounter the error that the access data of the WIFI on the board can no longer be changed.

The background is that I have two WIFI access points (lab and production environment).

If I store the respective WIFI credentials in the code and load the program on the board, the ESP32 still tries to establish a connection with the WIFI credentials previously entered in the code. However, the problem does not occur with the hostname or the SignalK server.

Is a problem of this kind already known or can someone tell me a way to completely reset an ESP32?

Translated with www.DeepL.com/Translator (free version)

Elzariant, 2022-05-25:
same issue for me. I use esptool to erase completely the data

mairas, 2022-05-27:
I'll reopen this because I think the raised point(s) are real issues and should be addressed.

That is:

1. Hardcoded WiFi credentials shouldn't be written to the file system (not sure if this is the case, needs to be verified).
2. There should be a button user interface for easier resetting of the WiFi configuration.

johannbarbie, 2023-01-17:
i've stumbled over this as well. for those searching for a quick-fix, the exact command to erase the flash is:

```
esptool.py --chip esp32 erase_flash
```

is there a link to the hardware button solution already?