SignalK/SensESP #624: RFE: Weather forecaster based on Zambretti Algorithm

mgrouch, 2022-08-24:
https://github.com/G6EJD/ESP32_Weather_Forecaster_TN061

http://www.wokinghamweather.co.uk/zambretti_algorithm_forecaster/zambretti_forecast.html

Thanks

ba58smith, 2022-08-24:
Sorry, @mgrouch - what is the issue?

tkurki, 2022-08-24:
Dumping random links without any explanation is very rarely helpful. What is it - you explain, dont ask maintainers to figure things out. Why should this be added? Why particularly in this repo? How would one use it? Who is it for?

mgrouch, 2022-08-24:
It's a local short time (one day) weather forecast based just on sensors you typically have on board.
Pressure, temperature, location and time, wind (just general direction and grade of strength).
Algorithm is very simple and results are 90% accurate or so.
One on-board can look at it around 9am (that's when it's most accurate) and plan the day accordingly.

ba58smith, 2022-08-24:
SensESP is almost exclusively used to collect data from sensors and get it to Signal K. The project you're describing sounds like it would be created as a Signal K plug-in, not something that SensESP would do.

mgrouch, 2022-08-24:
SignalK plugin would work too.

mairas, 2022-08-25:
It's a nice algorithm and potentially quite useful. A Signal K plugin would probably be the best approach.

@mgrouch, if you're interested in it, I'm afraid you need to implement it yourself.

Closing.

closed 2022-08-25 by mairas