SignalK/SensESP #634: SensESP stuck on old IP when SignalK changes IP

johannbarbie, 2023-01-17:
I restarted my PI after connecting the SensESP to my SignalK server and it got a new IP from DHCP.

now the SensESP is not able to connect, and I see the old IP in the SensESP interface:
[image]

First I was blaming it on SignalK to broadcast an outdated IP over mDNS, but using `avahi-resolve` in the network it correctly returns the new one `192.168.8.110`. 

So I assume the SensESP saves the value and doesn't update it. 

Obviously I can configure a fixed IP to the SignalK server, but wanted to leave the issue here as an improvement proposal.

My expectation would be that the SensESP rediscovers the correct IP when being restarted.

mairas, 2023-01-17:
If I understood correctly, you're using mDNS to discover the SK server, and SensESP "gets stuck" on the old IP, even after a restart?  That definitely sounds like a serious bug. Off the bat, I can't even think of a reason why SensESP should save the IP address in the first place. I'll look into it soon.

johannbarbie, 2023-01-17:
yes, I followed the [getting started tutorial](https://signalk.org/SensESP/pages/getting_started/) with the mDNS flag in SignalK enabled. 
I've used the 2.6.0 release of SensESP, and initiated it like this:

```
  SensESPAppBuilder builder;
  sensesp_app = builder.get_app();
  sensesp_app->start();
```

johannbarbie, 2023-01-17:
I see that the SignalK server is identified by `sk_server_address` and `sk_server_port` in SensESPApp class, which again is an instance of Configurable, and serializes these values to a file?

I guess instead of address and port, another identifier should be used, like hostname, which is then saved to the config when the access request to SignalK has been approved here: https://github.com/SignalK/SensESP/blob/cbd1ecdf0ec381e00908000e47d1e85a3ed6117b/src/sensesp/net/ws_client.cpp#L443-L446

This identifier/hostname would then be resolved to address+port on startup and then be used for `WSClient::connect()`.

mairas, 2023-01-19:
@johannbarbie could you test PR #636? The logic should be improved now.

johannbarbie, 2023-01-20:
yeah, that worked for me. 

1. started out with signalK on 192.168.178.177, authorized the sensor, and received data.
2. turned the sensor off, then assigned 192.168.178.201 to the machine of signalK 
3. restarted the sensor, and received data immediately.

I have few more observations:
funny enough, signalK and SensESP didn't start talking on 192.168.178.201 after the DHCP renewal, but using a link.local address 169.254.0.0/16. During that time I could see the IP of signalK on the status page, but the field on on configuration page was empty.
Then I used the configuration page to set the correct IP and port, then restarted the sensor. It worked and the sensor connected immediately. now I see the IP on the configuration page, but on the status page it is empty, and the port  wrong:

[image]

mairas, 2023-01-20:
OK, two things:

1. Who is Siggi and what did he do? (THIS IS IMPORTANT! I MUST KNOW!)
2. I'll have a look at the status page issue and amend this PR accordingly.

Note that if you set the IP address via the config page, that setting will preside even if the server gets a new address from the DHCP server. If the SK server address isn't fixed, it's better to leave the address in config empty, so that mDNS will be always used.

mairas, 2023-01-26:
LOL, glad I asked, and even more glad he's a former landlord!

Back to the topic, I just tested the status page issue and was able to replicate your results.

mairas, 2023-01-26:
Should be fixed now.

johannbarbie, 2023-01-26:
cool, thx for the effort. I'll be able to test it tomorrow.

johannbarbie, 2023-01-28:
I went through the full Sensor lifecycle with the latest commit and the `Signal K server address` and `Signal K server port` showed up correctly right from the beginning.

PR #636 (2023-02-14): Use mDNS always unless a server has been configured
The WSClient persistence logic now follows the rest of the codebase (I think...):

- If nothing is configured, mDNS is always used.
- If a server is hardcoded, it is used by default.
- If the server address is set through the config page, that address is used.
- If the server address is cleared through the config page, mDNS is used.

Fixes #634.
closed 2023-02-14 by mairas