SignalK/SensESP #644: RepeatSensor interval not configurable

avanlievenoogen, 2023-02-21:
Is there a way to make the interval configurable? Like for instance with the DigitalInputCounter read_delay?

ba58smith, 2023-02-21:
It is already configurable, in that you set the delay in the constructor:
`RepeatSensor<T>(unsigned int repeat_interval_ms, std::function<T()> callback)`.
If you mean you'd like it to be configurable at run time, from the Config UI, I'm sure that would be possible, but not sure how high a priority it would be, since you can just change it in `main.cpp`.

avanlievenoogen, 2023-02-22:
thanks for your response. I did mean configurable at run time and I agree, it is not really an issue. I just noticed it to be different than the other sensors and wanted to make sure I was not missing something.

closed 2023-02-22 by avanlievenoogen