SignalK/SensESP #653: upload OTA "No response from the ESP"

mat214, 2023-05-07:
Hello

I discover the esp32 and sensESP. I want to make a battery monitor.

The esp32 is connected in series to a victron mppt to recover its data.

An ina226 connected to a shunt must give the current.

I have a problem in the upload to the esp32 when the serial RX pin is connected. the plateformio upload is not done. I have to disconnect the pin for it to work.

So I would like to try the OTA upload. But that doesn't work either.

Do I have to write a particular code to have the OTA working on sensESP?

Here is what I have from plateformio

` pio run --target upload --upload-port 192.168.30.23`

```
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Uploading .pio\build\esp32dev\firmware.bin
10:25:47 [DEBUG]: Options: {'esp_ip': '192.168.30.23', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 47577, 'auth': '', 'image': '.pio\\build\\esp32dev\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
10:25:47 [INFO]: Starting on 0.0.0.0:47577
10:25:47 [INFO]: Upload size: 1532848
Sending invitation to 192.168.30.23 ..........
10:25:47 [ERROR]: No response from the ESP
```

EdKok, 2023-05-07:
Isn't this:
https://github.com/SensESP/ve.direct_mppt
way more accurate? Gives you the data already available in the MPPT and less clutter with hardware.

mat214, 2023-05-07:
I use the sensesp/VEDirect library

I failed to install the ve.direct_mppt

but the sensesp/VEDirect works, I manage to transfer the data to signalk. Although I haven't yet figured out how to use data in variables like vedi->parser.data.channel_1_battery_voltage
```

  vedi->parser.data.channel_1_battery_voltage.connect_to(new SKOutputFloat(
      "electrical.battery" SOLAR_CHARGE_CONTROLLER_ID ".voltage",
      "/Signal K/battery " SOLAR_CHARGE_CONTROLLER_ID " Voltage"));
```

I just happen with the example to send it to signalk. I would like to be able to use them in calculations specific to my program. Like for example switching the battery to full charge once in float.

It works but my main problem is that I can't upload a new version of my program if the rx pin I configured is not disconnected from the optocoupler...

mat214, 2023-05-07:
Sorry, I hadn't seen the page : https://signalk.org/SensESP/pages/features/

My OTA problem is solved. This method allows you to upload with the active connection between the esp32 and the mppt.

mat214, 2023-05-08:
Thanks to sensESP, I was able to advance my colometer project :
https://github.com/mat214/sensCoulo