SignalK/SensESP #666: Reset reason info of ESP32 available in SensESP/Signal K

KEGustafsson, 2023-08-25:
It would be good to have reset reason information available to SensESP UI or/and stream to Signal K.

Currently reset can be detected by monitoring USB debug stream or uptime data. If USB is not connected then it is very hard or imposdible to find out what was the reason of reset if it might happend.

ESP32 api support following debug info to be read from the system
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv416esp_reset_reasonv

JohnySeven, 2023-08-25:
Ok, 666 is a very devil 😈 issue! You can use UIOutput to show anything on the SensESP web UI home page.

KEGustafsson, 2023-08-27:
Noticed also devil sequence number, but I'm not superstitious 😉

I was thinking something like system_info_sensor reporting now and mainly to SK.

In SensESP Web UI, there would be nice to have also uptime info available in addition to last reset reason.

I could take a look to UIOutput. As C/C++ is not my core competencies, this might take some time, but I'm not in a hurry too.

JohnySeven, 2023-08-27:
UIOutput is simple concept, where you define the output (name, value, group), something like this:
`auto webLine = new UIOutput<String>("AIS Last sentence", "n/a", "NMEA");`

and then you call

`webLine->set('Hello world');`

So for simple reset reason web UI output it could look like:

IDF source: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html
At the top:

`#include <esp_system.h>`

In main:
```
  esp_reset_reason_t resetReason = esp_reset_reason();
  String textResetReason;

  switch(resetReason)
  {
    case ESP_RST_UNKNOWN:
      textResetReason = "Unknown";
      break;
    case ESP_RST_POWERON:
      textResetReason = "Power on";
      break;
    case ESP_RST_EXT:
      textResetReason = "External";
      break;
    case ESP_RST_SW:
      textResetReason = "Software";
      break;
    case ESP_RST_PANIC:
      textResetReason = "Panic";
      break;
    case ESP_RST_INT_WDT:
      textResetReason = "Internal Watchdog";
      break;
    case ESP_RST_TASK_WDT:
      textResetReason = "Task Watchdog";
      break;
    case ESP_RST_WDT:
      textResetReason = "Watchdog";
      break;
    case ESP_RST_DEEPSLEEP:
      textResetReason = "Deep Sleep";
      break;
    case ESP_RST_BROWNOUT:
      textResetReason = "Brownout";
      break;
    case ESP_RST_SDIO:
      textResetReason = "SDIO";
      break;
  }

  auto webResetReason = new UIOutput<String>("Reset Reason", textResetReason, "System");
```