SignalK/SensESP #668: Fixed repeat times at system info sensors

ckcrs, 2023-10-26:
Hi,
...src/sensesp/sensors/system_info.cpp has hard coded milisecond values for repeat loops and it is stated:
" Nothing about it is configurable, except whether or not it appears in your project's output. That is configured with SensESPAppBuilder."
I wonder the reasoning, for example the ip address is emitted at every 10 secs, any reason not having a longer time, **even why they are not configurable during activation**? In my case, once the two esp32 has got ip address they seems not changing, so  even one minute emit time would be quite ok.
Thanks,

mairas, 2023-10-27:
I don't think there's strong reasoning. I'm not a huge fan of transmitting the device IP address over SK because it feels rather redundant, but other people have wanted it. You're right that the delay could be longer.

ckcrs, 2023-10-27:
Thanks.
 I've adjusted ...src/sensesp/sensors/system_info.cpp file content as:
void IPAddrDev::start() {
  ReactESP::app->onRepeat(61000, [this]() { this->update(); }); // original delay time: 10000 
}
Now it suits my case, hence closing issue.

closed 2023-10-27 by ckcrs