SignalK/SensESP #688: Bug with FloatSKListener() in Dev-3

mestafin, 2024-04-15:
To verify a potential bug in a larger sensESP program using the DEV-3 branch, I created a small senseESP test program with one line (subscribing to output of SKSimulator running on Signal K version 2.71);

 `auto *sk_rpm_hz = new FloatSKListener("propulsion.p0.revolutions", 1000);`

The program crashes immediately after the subscription was submitted to Signal K. The error log displayed on the console is given below.

The program works 100% when compiled with version 2.7.2 of senseESP

```
.....
(websocket_event_handler)(C1) WEBSOCKET_EVENT_CONNECTED
(on_connected)(C1) Subscribing to Signal K listeners...
(subscribe_listeners)(C1) Adding propulsion.p0.revolutions subscription with listen_delay 1000

(subscribe_listeners)(C1) Subscription JSON message:
 {"context":"vessels.self","subscribe":[{"path":"propulsion.p0.revolutions","period":1000}]}
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2f81  PS      : 0x00060930  A0      : 0x800d3aad  A1      : 0x3ffb20b0  
A2      : 0x00000000  A3      : 0x00000007  A4      : 0x00060120  A5      : 0x00060123  
A6      : 0x00060123  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffdcd94  
A10     : 0x3ffc9f10  A11     : 0x3ffd3534  A12     : 0x00060120  A13     : 0x00060123  
A14     : 0x007bf588  A15     : 0x003fffff  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x40084699  LEND    : 0x400846a1  LCOUNT  : 0x00000027  

Backtrace: 0x400d2f7e:0x3ffb20b0 0x400d3aaa:0x3ffb20d0 0x400d972d:0x3ffb2110 0x400e2705:0x3ffb2140 0x400e2768:0x3ffb2170 0x400e27fd:0x3ffb21c0 0x400d401e:0x3ffb21e0 0x400d41a9:0x3ffb2200 0x400d42c9:0x3ffb2230 0x400d42e1:0x3ffb2250 0x400d3316:0x3ffb2270 0x400e91a1:0x3ffb2290

  #0  0x400d2f7e:0x3ffb20b0 in ArduinoJson::V704PB2::detail::VariantPool::getSlot(unsigned short) const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Memory/VariantPoolImpl.hpp:49
      (inlined by) ArduinoJson::V704PB2::detail::VariantPoolList::getSlot(unsigned short) const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Memory/VariantPoolList.hpp:99
  #1  0x400d3aaa:0x3ffb20d0 in ArduinoJson::V704PB2::detail::ResourceManager::getSlot(unsigned short) const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Memory/ResourceManager.hpp:64
      (inlined by) ArduinoJson::V704PB2::detail::CollectionData::createIterator(ArduinoJson::V704PB2::detail::ResourceManager const*) const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Collection/CollectionData.hpp:88
      (inlined by) ArduinoJson::V704PB2::detail::CollectionIterator ArduinoJson::V704PB2::detail::ObjectData::findKey<ArduinoJson::V704PB2::detail::StaticStringAdapter>(ArduinoJson::V704PB2::detail::StaticStringAdapter, ArduinoJson::V704PB2::detail::ResourceManager const*) const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Object/ObjectImpl.hpp:32
  #2  0x400d972d:0x3ffb2110 in ArduinoJson::V704PB2::detail::VariantData* ArduinoJson::V704PB2::detail::ObjectData::getMember<ArduinoJson::V704PB2::detail::StaticStringAdapter>(ArduinoJson::V704PB2::detail::StaticStringAdapter, ArduinoJson::V704PB2::detail::ResourceManager const*) const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Object/ObjectImpl.hpp:15
      (inlined by) ArduinoJson::V704PB2::detail::VariantData* ArduinoJson::V704PB2::detail::ObjectData::getMember<ArduinoJson::V704PB2::detail::StaticStringAdapter>(ArduinoJson::V704PB2::detail::ObjectData const*, ArduinoJson::V704PB2::detail::StaticStringAdapter, ArduinoJson::V704PB2::detail::ResourceManager const*) at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Object/ObjectData.hpp:54
      (inlined by) ArduinoJson::V704PB2::detail::VariantData* ArduinoJson::V704PB2::detail::VariantData::getMember<ArduinoJson::V704PB2::detail::StaticStringAdapter>(ArduinoJson::V704PB2::detail::StaticStringAdapter, ArduinoJson::V704PB2::detail::ResourceManager const*) const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:201
      (inlined by) ArduinoJson::V704PB2::detail::VariantData* ArduinoJson::V704PB2::detail::VariantData::getMember<ArduinoJson::V704PB2::detail::StaticStringAdapter>(ArduinoJson::V704PB2::detail::VariantData const*, ArduinoJson::V704PB2::detail::StaticStringAdapter, ArduinoJson::V704PB2::detail::ResourceManager const*) at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:209
      (inlined by) ArduinoJson::V704PB2::detail::MemberProxy<ArduinoJson::V704PB2::JsonObject, char const*>::getData() const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:49
  #3  0x400e2705:0x3ffb2140 in decltype (({parm#1}.getData)()) ArduinoJson::V704PB2::detail::VariantAttorney::getData<ArduinoJson::V704PB2::detail::MemberProxy<ArduinoJson::V704PB2::JsonObject, char const*> const>(ArduinoJson::V704PB2::detail::MemberProxy<ArduinoJson::V704PB2::JsonObject, char const*> const&) at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantAttorney.hpp:26
      (inlined by) ArduinoJson::V704PB2::detail::VariantRefBase<ArduinoJson::V704PB2::detail::MemberProxy<ArduinoJson::V704PB2::JsonObject, char const*> >::getData() const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:260
      (inlined by) ArduinoJson::V704PB2::detail::VariantRefBase<ArduinoJson::V704PB2::detail::MemberProxy<ArduinoJson::V704PB2::JsonObject, char const*> >::getVariantConst() const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:270
      (inlined by) ArduinoJson::V704PB2::detail::enable_if<!ArduinoJson::V704PB2::detail::ConverterNeedsWriteableRef<char const*>::value, char const*>::type ArduinoJson::V704PB2::detail::VariantRefBase<ArduinoJson::V704PB2::detail::MemberProxy<ArduinoJson::V704PB2::JsonObject, char const*> >::as<char const*>() const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:52
  #4  0x400e2768:0x3ffb2170 in ArduinoJson::V704PB2::detail::VariantRefBase<ArduinoJson::V704PB2::detail::MemberProxy<ArduinoJson::V704PB2::JsonObject, char const*> >::operator char const*<char const*, void>() const at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:63
      (inlined by) sensesp::WSClient::process_received_updates() at .pio/libdeps/esp32dev/SensESP/src/sensesp/net/ws_client.cpp:284
  #5  0x400e27fd:0x3ffb21c0 in std::_Function_handler<void (), sensesp::WSClient::WSClient(String, sensesp::SKDeltaQueue*, String, unsigned short, bool)::{lambda()#2}>::_M_invoke(std::_Any_data const&) at .pio/libdeps/esp32dev/SensESP/src/sensesp/net/ws_client.cpp:92
      (inlined by) _M_invoke at /Users/johan/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #6  0x400d401e:0x3ffb21e0 in std::function<void ()>::operator()() const at /Users/johan/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
  #7  0x400d41a9:0x3ffb2200 in reactesp::RepeatReaction::tick() at .pio/libdeps/esp32dev/ReactESP/src/ReactESP.cpp:66
  #8  0x400d42c9:0x3ffb2230 in reactesp::ReactESP::tickTimed() at .pio/libdeps/esp32dev/ReactESP/src/ReactESP.cpp:151
  #9  0x400d42e1:0x3ffb2250 in reactesp::ReactESP::tick() at .pio/libdeps/esp32dev/ReactESP/src/ReactESP.cpp:166
  #10 0x400d3316:0x3ffb2270 in loop() at src/main.cpp:47
  #11 0x400e91a1:0x3ffb2290 in loopTask(void*) at /Users/johan/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 1a0f25bd44436a3b

Rebooting...
```

mairas, 2024-04-16:
Thanks a lot for the report. I have updated ArduinoJson from v6 to v7 and some assignment semantics had apparently changed. Fixed now.

mairas, 2024-04-16:
Meh, auto-closing didn't work.

closed 2024-04-16 by mairas