SignalK/SensESP #698: Unity and delay

Champang, 2024-05-24:
I tested a BNO085, 9-axis IMU to send via WiFi with a Lolin D32 on signalK on a Raspberry 3+.
Problems
- in angle units. OpenCPN does not read radians, I have to specify "navigation.headingCompass","units","deg".
- the period between two acquisitions with a value less than 500 milliseconds is versatile.
`
/* 
 * Sensesp : https://github.com/SignalK/SensESP
 * send data to 
 * signalk : https://github.com/SignalK/signalk-server
 * BNO085 IMU 9 axes exemple : https://www.adafruit.com/product/4754
 * This example outputs Heading, Roll, Pitch every 500 ms
 * Use Visual Code Studio and Platformio
 * see PlatformIO Project Configuration File at the bottom of the script.
*/

// Sensor-specific #includes:
#include <Arduino.h>
#include <Wire.h>
#include <math.h>
#include <SparkFun_BNO080_Arduino_Library.h>  // Device I2C = 0x4B if DI = Vcc

// Boilerplate #includes:
#include "sensesp/sensors/sensor.h"
#include "sensesp/signalk/signalk_output.h"
#include "sensesp_app.h"
#include "sensesp_app_builder.h"

#define I2C_BUFFER_LENGTH 32
//#define I2C_SCL 	22
//#define I2C_SDA 	21

using namespace sensesp;

reactesp::ReactESP app;
BNO080 myIMU;

float degree = 2;
float pitch = 2;
float roll = 1;
float yaw = 1;

float yaw_callback() {
  if (myIMU.dataAvailable() == true)
  {    
    yaw = (myIMU.getYaw()) * 180.0 / PI; // Convert yaw / navigation.heading to degrees
    if (yaw < 0) yaw += 360.0;
    degree = 360 - yaw;
    if (degree < 0) degree += 360.0;
    }
    int temp = degree;
    yaw = temp;
    
    //yaw = (myIMU.getYaw());
  return (yaw);
}

float pitch_callback() {
  if (myIMU.dataAvailable() == true)
  {
    pitch = (myIMU.getPitch() * 180.0 / PI); // Convert pitch to degrees
  }
  int temp = pitch;
  pitch = temp;
  return (pitch);
}

float roll_callback() {
  if (myIMU.dataAvailable() == true)
  {
    roll = (myIMU.getRoll() * 180.0 / PI); // Convert roll to degrees
  }
  int temp = roll;
  roll = temp;
  return (roll);
}

void setup() {
  // Some initialization boilerplate when in debug mode...
#ifndef SERIAL_DEBUG_DISABLED
  SetupSerialDebug(115200);
#endif

  Wire.begin(21, 22);
  Wire.setClock(400000);
  delay(1000);
  myIMU.begin();
  myIMU.enableRotationVector(30);

  // Construct the global SensESPApp() object
  SensESPAppBuilder builder;

  sensesp_app = (&builder)
                    // Set a custom hostname for the app.
                    ->set_hostname("YOUR_HOSTNAME")
                    // Optionally, hard-code the WiFi and Signal K server
                    // settings. This is normally not needed.
                    ->set_wifi("SSID", "PASSWORD")
                    ->set_sk_server("IP_SignalK_server", 3000)
                    ->get_app();

  // Create the ConstantSensor object all 500 millisecondes
  auto* yaw = new RepeatSensor<float>(500, yaw_callback);
  auto* pitch = new RepeatSensor<float>(500, pitch_callback);
  auto* roll = new RepeatSensor<float>(500, roll_callback);

  // create and connect the output object
  yaw->connect_to(new SKOutputFloat("navigation.headingCompass","units","deg"));
  pitch->connect_to(new SKOutputFloat("navigation.attitudepitch","units","deg"));
  roll->connect_to(new SKOutputFloat("navigation.attituderoll","units","deg"));

  // Start the SensESP application running
  sensesp_app->start();
}

void loop() { app.tick(); }

/*
; PlatformIO Project Configuration File : Lolin D32
;  Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:lolin_d32]
platform = espressif32
board = lolin_d32
framework = arduino
board_build.mcu = esp32
board_build.f_cpu = 240000000L
board_build.flash_size = 4MB
board_build.flash_mode = dio
board_build.partitions = min_spiffs.csv
lib_deps = 
	;adafruit/Adafruit Unified Sensor
	;sparkfun/SparkFun BNO080 Cortex Based IMU
	;signalk/SensESP
	adafruit/Adafruit Unified Sensor @ ^1.1.14
	sparkfun/SparkFun BNO080 Cortex Based IMU @ ^1.1.12
	signalk/SensESP@^2.7.2
;upload_port = /dev/ttyUSB0
;upload_flags =
;    --upload-reset
*/
`

Champang, 2024-05-25:
Hello, you confirm the weakness of OpenCPN in unit management.
Concerning the latency of data exchanges, my objective is to be able to recover certain data on an Esp8266 to display them on a Nokia 5110.
I don't know how to directly query Signalk-server to retrieve this data, so I use Mosquitto.
When I query signalk-server on http://10.10.10.1:3000/admin/#/databrowser, I notice that if I put a value less than 500ms in sending data, the display is less responsive following the displacement of BNO085.
`
  // Create the ConstantSensor object all 500 millisecondes
  auto* yaw = new RepeatSensor<float>(500, yaw_callback);
  auto* pitch = new RepeatSensor<float>(500, pitch_callback);
  auto* roll = new RepeatSensor<float>(500, roll_callback);`

mairas, 2024-05-25:
Repeat reactions can run tens of thousands of times per second without any issues. If you're having latency issues, it's likely that your accelerometer library functions are blocking for a long time.

I'm closing this issue because these problems do not seem to be related to SensESP itself.

closed 2024-05-25 by mairas