SignalK/SensESP #729: Not saving wifi client credentials in setup screen.

imoldovavan, 2024-08-17:
In wifi setup screen there is no indication in the output terminal that the credentials are saved. Hitting the save button does nothing. If I switch back and forth between SignalK tab and WiFi the credentials disappear.

imoldovavan, 2024-08-18:
If I connect the pi to the esp32 ap, everything seems to be working. Not sure if this info is of any help :)

donmarlin, 2024-08-23:
I can confirm this. Using the Beta version. All was working well after upgrading from Alpha. 
I then did factory reset using the web UI to clear the nvram.  
Now wifi wont connect to the network. I have tried to set the wifi credentials in the UI but they don't seem to be saved. 
I also tried to hardcode wifi in builder config but nothing seems to work.

grhitch, 2024-08-24:
I can also confirm this problem using SensESP 3.0.0-beta.1.
Hardware is various, but includes Halmet v1.0.

1) Hard coded wifi settings are not effective, although changing the hostname here does work.
2) Web interface scans wifi and finds access points, but cannot connect and does not save any changes. 

I have tried to get to the bottom of this, but unfortunately do not have the skills  to make any real progress. 

I did manage to confirm that SPIFFS does exist. I also confirmed that a file is created when starting with a 'blank' device. I added a couple of debug lines that yielded:
> FILE: qLsPBGc6u2S+SPgr87i11VrDlCE=
> File size: 31 bytes

Watching the HTTP messages didn't show anything that was obviously "save".

An abbreviated log file is attached

[device-monitor-240824-125808.log](https://github.com/user-attachments/files/16735003/device-monitor-240824-125808.log)

mairas, 2024-08-25:
I was able to reproduce this and just fixed the problem. I had made some last-minute changes in the frontend after which I got distracted and didn't test them properly.

Until I release a new beta version, you can use the latest development version by commenting out the SensESP dependency in your project `platformio.ini` file and adding the following:

```
https://github.com/SignalK/SensESP.git
```

closed 2024-08-25 by mairas