SignalK/SensESP #730: Compile error in analog_reader.h with S2 and S3 variants

CaptIgmu, 2024-08-18:
Getting compile error in platformio with SensESP v3 in file **_analog_reader.h_** when using ESP32-S2 and S3 variants:

**_pio/libdeps/adafruit_qtpy_esp32s3_n4r2/SensESP/src/sensesp/sensors/analog_reader.h:28:30: error: 'ADC_ATTEN_DB_12' was not declared in this scope
   adc_atten_t attenuation_ = ADC_ATTEN_DB_12;
                              ^~~~~~~~~~~~~~~
.pio/libdeps/adafruit_qtpy_esp32s3_n4r2/SensESP/src/sensesp/sensors/analog_reader.h:28:30: note: suggested alternative: 'ADC_ATTEN_DB_11'_**

I can get it to compile by changing to  ADC_ATTEN_DB_11 but noticed in the ESP-IDF docs:
**_enumerator ADC_ATTEN_DB_11
    This is deprecated, it behaves the same as ADC_ATTEN_DB_12_**

So not sure if a **_Pull request_** will work for all ESP32s !

imoldovavan, 2024-08-18:
Try ADC_ATTEN_DB_0

mairas, 2024-08-21:
Could you provide your `platformio.ini` file? I tried with setting the board to `esp32-s3-devkitc-1` and used a simple program with the following:

```
  auto* analog_input =
      new AnalogInput(pin, read_delay, analog_in_config_path, output_scale);

  auto analog_sk_output = new SKOutput<float>(sk_path);
  analog_input->connect_to(analog_sk_output);
```

With the above setup, I couldn't reproduce the error.

CaptIgmu, 2024-08-23:
Here is my **_platformio.ini_** file:

[env]
lib_ldf_mode = deep
monitor_speed = 115200
upload_speed = 460800

[espressif32_base]
build_unflags = -Werror=reorder
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder

[env:adafruit_qtpy_esp32s3_n4r2]
extends = espressif32_base
platform = espressif32
board = adafruit_qtpy_esp32s3_n4r2
framework = arduino
lib_deps = 
	signalk/SensESP @ >=3.0.0-beta.1,<4
	# https://github.com/SignalK/SensESP.git  # Use this line to use the latest git version
  	# symlink:///Users/mairas/src/SignalK/SensESP  # Use this line to use a local copy
	mairas/ReactESP @ >=2.1.0,<3
	sensesp/OneWire @ >=3.0.0,<4

build_flags =
	; Max (and default) debugging level in Arduino ESP32 Core
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    ; Arduino Core bug workaround: define the log tag for the Arduino
    ; logging macros.
    -D TAG='"Arduino"'
    ; Use the ESP-IDF logging library - required by SensESP.
    -D USE_ESP_IDF_LOG

janifr, 2024-09-06:
I have the exact same problem compiling unmodified project template.

`In file included from .pio/libdeps/esp32dev/SensESP/src/sensesp/sensors/analog_input.h:4,
                 from src/main.cpp:11:
.pio/libdeps/esp32dev/SensESP/src/sensesp/sensors/analog_reader.h:28:30: error: 'ADC_ATTEN_DB_12' was not declared in this scope
   adc_atten_t attenuation_ = ADC_ATTEN_DB_12;
                              ^~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/SensESP/src/sensesp/sensors/analog_reader.h:28:30: note: suggested alternative: 'ADC_ATTEN_DB_11'
   adc_atten_t attenuation_ = ADC_ATTEN_DB_12;
                              ^~~~~~~~~~~~~~~
                              ADC_ATTEN_DB_11`

CaptIgmu, 2024-09-09:
Same error with my new **_Halmet_** board with regular ESP32!

GHBLoos, 2024-09-11:
Same issue. I didn't change anything of the downloaded code.
After changing it to `ADC_ATTEN_DB_11;`, the build succeeds

smeisner, 2024-09-15:
I am seeing this as well with an ESP32-DevKitC board. I will eventually be using the ESP32-S3, so it won't be resolved with the target HW.

If I change it to ADC_ATTEN_DB_11 or ADC_ATTEN_DB_0, it works fine.

Is there a problem changing this to ADC_ATTEN_DB_0 (as you commented above) in the source file?

JohnySeven, 2024-11-22:
Hit the same roadblock.

Steps to reproduce:
1. Created SensESP repo from template https://github.com/SensESP/SensESP-project-template
2. Git clone to my computer and opened VS Code
3. Hit that build button

Same issue.

mairas, 2024-11-25:
> Is it possible you have a custom sdkconfig file for your ESP controller? I'm wondering if a setting in there can cause your defaults to change.

No, not really. When using the default Arduino framework, sdkconfig.h is always provided by the framework (unless you very specifically configure Arduino as an ESP-IDF component, but that's a whole different kettle of fish).

GHBLoos, 2024-11-25:
I just build a previous project -> Build succeeded
I removed .pio folder -> Build succeeded
I updated the SensESP library from 3.0b6 to 3.0. Now I get the following error when building

```
In file included from .pio/libdeps/esp32dev/SensESP/src/sensesp/sensors/analog_input.h:5,
                 from src/main.cpp:11:
.pio/libdeps/esp32dev/SensESP/src/sensesp/sensors/analog_reader.h:28:30: error: 'ADC_ATTEN_DB_12' was not declared in this scope
   adc_atten_t attenuation_ = ADC_ATTEN_DB_12;
                              ^~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/SensESP/src/sensesp/sensors/analog_reader.h:28:30: note: suggested alternative: 'ADC_ATTEN_DB_11'
   adc_atten_t attenuation_ = ADC_ATTEN_DB_12;
                              ^~~~~~~~~~~~~~~
                              ADC_ATTEN_DB_11
``` 

Platformio.ini
```PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_ldf_mode = deep
upload_speed = 460800
monitor_speed = 115200
lib_deps = 
	sensesp/OneWire@^3.0.1
	signalk/SensESP@^3.0.0
build_unflags = 
	-Werror=reorder
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder
extends = espressif32_base
build_flags = 
	-D LED_BUILTIN=2
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
	-D TAG='"Arduino"'
	-D USE_ESP_IDF_LOG
```

GHBLoos, 2024-11-25:
Do you mean by renaming .platformio directory the .pio directory?

JohnySeven, 2024-11-26:
> If/when someone tries to clean their `.platformio` directory, it might be a good idea to just rename it. If that fixes the issue, the exact version changes could be investigated.

Will try to clean my platformio and see how that goes. Little busy with life right now, sorry!

GHBLoos, 2024-11-26:
I am not a real coder, so I just follow the instructions I find on the SensESP github site/Signalk site or whatever I find using Google to set up stuff. From the moment it runs I can work with the code. So below I just drop whatever comes up on my screen. Hopefully it gives a lead to the issue.

Renamed the .platformio directory. A new directory is created.
After starting VStudio the Output window shows
```
[11/26/2024, 5:50:17 PM] Unable to resolve configuration with compilerPath "C:/Users/<USER>/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc.exe".  Using "cl.exe" instead.
[11/26/2024, 5:50:17 PM] Unable to resolve configuration with compilerPath "C:/Users/<USER>/.platformio/packages/toolchain-xtensa-esp32s2/bin/xtensa-esp32s2-elf-gcc.exe".  Using "cl.exe" instead.
```
Opening project and build
```
Compiling .pio\build\esp32dev\lib292\OneWireNg\platform\Platform_Delay.cpp.o
esptool.py v4.5.1
In file included from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_TimeCritical.h:18,
Creating esp32 image...
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_Delay.h:16,
Merged 1 ELF section
                 from .pio/libdeps/esp32dev/OneWireNg/src/drivers/DSTherm.cpp:14:
Successfully created esp32 image.
C:/Users/gaston/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_TimeCritical.h:18,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_Delay.h:16,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/OneWireNg_ArduinoIdfESP32.h:17,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/OneWireNg_ArduinoIdfESP32.cpp:13:
C:/Users/gaston/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
In file included from src/main.cpp:1:
C:/Users/gaston/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_TimeCritical.h:18,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_Delay.h:16,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_Delay.cpp:13:
C:/Users/gaston/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_TimeCritical.h:18,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_Delay.h:16,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/OneWireNg_ArduinoIdfESP8266.h:17,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/OneWireNg_ArduinoIdfESP8266.cpp:13:
C:/Users/gaston/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_TimeCritical.h:18,
                 from .pio/libdeps/esp32dev/OneWireNg/src/platform/Platform_Delay.h:16,
                 from .pio/libdeps/esp32dev/OneWireNg/src/OneWireNg_BitBang.cpp:14:
C:/Users/gaston/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
*** [.pio\build\esp32dev\lib292\OneWireNg\drivers\DSTherm.cpp.o] Error 1
*** [.pio\build\esp32dev\lib292\OneWireNg\platform\Platform_Delay.cpp.o] Error 1
*** [.pio\build\esp32dev\lib292\OneWireNg\platform\OneWireNg_ArduinoIdfESP32.cpp.o] Error 1
*** [.pio\build\esp32dev\lib292\OneWireNg\OneWireNg_BitBang.cpp.o] Error 1
*** [.pio\build\esp32dev\lib292\OneWireNg\platform\OneWireNg_ArduinoIdfESP8266.cpp.o] Error 1
============================================================================= [FAILED] Took 6.82 seconds =============================================================================
 *  The terminal process "C:\Users\gaston\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 ```
 Not sure what lib FreeRTOS is and why it is not loaded.

Does this have anything to do with the issue (before I try to solve it myself)?