SignalK/SensESP #738: Migrate to ESP-IDF 5.x and Arduino 3.x

mairas, 2024-08-30:
The Espressif and PlatformIO are in an unfortunate [staring contest](https://github.com/platformio/platform-espressif32/issues/1225) wrt. supporting Arduino Core 3 and ESP-IDF 5.x in PlatformIO. As a result, the ESP32 Arduino Core in PlatformIO isn't getting updates or bugfixes which I would like to use in SensESP. Given that a new [pioarduino](https://github.com/pioarduino/platform-espressif32) project has emerged to bridge this gap, SensESP should probably adopt it. Using the updated platform requires only a one-liner change in `plaformio.ini` for the end user, but the changes in the framework itself require some trivial fixes in SensESP itself.

I don't yet know whether it is feasible to try to support both Arduino core versions simultaneously.

mefenlon, 2025-01-12:
I think this is related to a problem I reported over at https://github.com/SensESP/SensESP-project-template/issues/6#issue-2782472280

A few days ago, I started getting a build error

> fatal error: esp_websocket_client.h: No such file or directory

It looks like some issues have come up for  platformio
https://github.com/platformio/platform-espressif32/issues/1225#issuecomment-2578821334

CaptIgmu, 2025-03-09:
With migration to **_pioarduino_** to circumvent PlatformIO troubles, there is now a compile problem:
**_fatal error: esp_websocket_client.h: No such file or directory_**

That is because in ESP-IDF 5.x, **_esp_websocket_client.h_** has moved to **_espressif/esp-protocols/components/esp_websocket_client/include_**

Trying to "point at it" with include directories or adding the necessary files to a project src folder didn't work for me. There will have to be a fundamental SensESP change.

Mairas, Thanks for all your work!

mefenlon, 2025-03-10:
@CaptIgmu you can use this workaround.
Use this for your project as a template: https://github.com/SignalK/SensESP/blob/main/examples/platformio.ini

Chgange lib_deps to use the git repository rather than the release

```
lib_deps =
  https://github.com/SignalK/SensESP.git
```

CaptIgmu, 2025-03-11:
Wow! **_That worked!_** 
Looks like major changes in the platformio.ini file for SensESP.

I was still getting compile errors in my program using SensESP/OneWire, had to update my program based on the latest example:
**_using namespace reactesp;
using namespace sensesp;
using namespace sensesp::onewire;_**

Thanks for your help, I can now explore my Halmet, but now I'm intrigued about the **_Halser . . ._**

mefenlon, 2025-05-23:
@ErikDLundberg sorry for the late reply. I am no expert on this, but I have some thoughts on your issue.

Version 3.1.0 was released so my workaround to set _lib_deps_ is no longer necessary.

I would suggest getting a fresh copy of the current platform.ini file from the current repo https://github.com/SignalK/SensESP/blob/main/platformio.ini

The only change you should have to make in the platform.io file is setting the environment at the top of the file
```
; Change the default environment to match your board. Valid options are:
; - arduino_esp32
; - pioarduino_esp32
; - espidf_esp32
; - pioarduino_esp32c3
; - espidf_esp32c3
; - shesp32
; - halmet
; - halser

default_envs = pioarduino_esp32
```

I don't have any experience with the FireBeetle 2.  I think you could get away at leaving it at the default value unless you know for sure which env the firebeatle is.

I use a halmet so as an example I would change that one line to 
```
; Change the default environment to match your board. Valid options are:
; - arduino_esp32
; - pioarduino_esp32
; - espidf_esp32
; - pioarduino_esp32c3
; - espidf_esp32c3
; - shesp32
; - halmet
; - halser

default_envs = halmet
```

It sounds like you might have some dependency issues, so start from an empty directory and a new project just to make sure you are starting clean.