SignalK/SensESP #741: 3.0.0 beta 2 Issue: connection to SK stops after ~9200 deltas

nadrog, 2024-09-14:
Hi,

  just updated to 3.00 beta 2 and consistently, the deltas are not posted anymore after approx 9200 messages sent (which takes an hour and 12 minutes - same duration on two repeats). After reset, everything works (with described issue).

  This is on SailorHat.

Br, G

donmarlin, 2024-09-19:
I can confirm that the same thing happens to my system, I am running the .Beta-2 release. After about 72 minutes SK messages stop updating to the server. I can still manage Sensesp thru the web UI and when I restart Sensesp everything starts running normally. The logging output stops completely. In my case I restarted the node at 18:06 and SK messages stopped at 19:18 which is 72 minutes. I captured all the debug logs and there were no warnings or errors in the logs prior to communications stopping.

nadrog, 2024-10-07:
Looks like resolved in beta 3