SignalK/SensESP #752: OTA upload issue

Haimana, 2024-10-01:
Hello,

I'm trying to upload with OTA but I encounter some errors and I'm stuck. 
If direct upload (through com port, windows) everything is just fine.

```
Uploading .pio\build\esp32dev\firmware.bin
13:58:07 [DEBUG]: Options: {'esp_ip': '192.168.1.177', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 17309, 'auth': 
'password', 'image': '.pio\\build\\esp32dev\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
13:58:07 [INFO]: Starting on 0.0.0.0:17309
13:58:07 [INFO]: Upload size: 1293888
Sending invitation to 192.168.1.177 
Authenticating...OK
13:58:08 [INFO]: Waiting for device...
13:58:18 [ERROR]: No response from device
*** [upload] Error 1
```

In main.cpp I have this lines for OTA:

```
  sensesp_app = (&builder)
                    // Set a custom hostname for the app.
                    ->enable_ota("password")
                    ->get_app();

```
And platformio.ini (with uncommented OTA lines)

```
; PlatformIO Project Configuration File
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
;set default_envs to whichever board(s) you use. Build/Run/etc processes those envs
default_envs =
  esp32dev

[env]
; Global data for all [env:***]
framework = arduino
lib_ldf_mode = deep
monitor_speed = 115200
lib_deps =
  ; Peg the SensESP version to 2.0.0 and compatible versions
  SignalK/SensESP @ ^2.7.2
  ; Add any additional dependencies here
	SensESP/OneWire @ ^2.0.0

[espressif32_base]
;this section has config items common to all ESP32 boards
platform = espressif32
build_unflags =
  -Werror=reorder
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder

[env:esp32dev]
extends = espressif32_base
board = esp32dev
build_flags =
  -D LED_BUILTIN=2
  ; Uncomment the following to disable debug output altogether
  ;-D DEBUG_DISABLED
  ; Uncomment the following to enable the remote debug telnet interface on port 23
  ;-D REMOTE_DEBUG

;; Uncomment the following lines to use Over-the-air (OTA) Updates
upload_protocol = espota
upload_port = "192.168.1.177"
upload_flags =
  --auth=password

```

Any advice will be appreciated

Thank you

mairas, 2024-10-01:
Your setup would seem to be spot on, no issues there. Also, it seems like something goes through because of the "Authenticating...OK" line.

Could it be due to firewall software? See this discussion: https://community.platformio.org/t/problems-with-esp32-ota-programming/19198/6

Haimana, 2024-10-01:
It seems I need a new Windows...
It works OK on different computer.
On mine, still not working, not even after uninstall, delete .vscode and .platformio folders and reinstall everything
I'll close this for now.
Thank you for support and sorry for false alarm

closed 2024-10-01 by Haimana

Haimana, 2024-10-02:
fixed by adding

upload_flags =
  --host_port=45678

in platformio.ini
and inbound and outboud rules for this port (TCP and UDP) in windows firewall