SignalK/SensESP #789: Issues overloading set() function on custom ValueConsumer for a display

caballero03, 2024-11-16:
I have been experimenting with SensESP v3.0 and mostly having great success. However, as I try to do more advanced things I have run into a snag. What I am trying to do (maybe mistakenly) is have a custom ValueConsumer class that when provided a float value will display the numeric value or graphical representation of the value on the display and when provided with an integer value, it will apply that to the LCD brightness.

I have all of this working with two separate classes; one for displaying values and one for backlight brightness control. My initial design was to combine these in a way that might make it possible to change the display color to red when the ambient light level is low in addition to dimming the LCD backlight.

Can anyone shed some light on what I might need to do here? TIA

Here are some snippets of my code and error messages that I am getting when trying to make a custom polymorphic ValueConsumer class:

```C++
  // MyDisplay.h class definition
  
  class MyDisplay  : public FloatConsumer, public FloatProducer//, public IntConsumer <--- Cannot get IntConsumer to work without breaking things
                     {
	public:
		// Constructors
		MyDisplay( ...);

		MyDisplay( ...);

		virtual void set(const float& new_value) override;
		virtual void set(const int& new_value); // <-- Can't override without IntConsumer (of course)
		
	... other methods ...
  }

  // Usage of class in main.cpp and errors encountered

  const char* DBT_sk_path = "environment.depth.belowTransducer";

  /////////////////////////////////////////////
  // Depth below transducer
  auto* depthListener = new FloatSKListener(DBT_sk_path);

  depthListener
    ->connect_to(new MyDisplay(&tft, ROW_1, COL_3, "m", FONT_2)) // <--- connect_to() fails with "no instance of overloaded function "sensesp::SKValueListener<T>::connect_to [with T=float]" matches the argument list" when IntConsumer is added to class header
    // Convert meters to feet
    ->connect_to(new Linear(METERS_TO_FEET, 0.0))
    ->connect_to(new MyDisplay(&tft, ROW_5, COL_3, "ft", FONT_2))
    ->connect_to(new MyDisplay(&tft, DisplayType::VBAR, ROW_13, COL_10, 100, "ft", FONT_2));
```

mairas, 2024-11-16:
Having a single class inheriting from multiple consumer types was a pattern used in previous SensESP versions, but I removed support for it during the version 3 development. I believe some technical reasons (that I fail to remember) were the breaking point, but I also didn't like the pattern much. It wasn't possible to create multiple consumers of the same type, and the usage of such multi-consumers could be confusing.

However, there's a better pattern available. In `MyDisplay`, define public `LambdaConsumer` members for the inputs you need. For example, for brightness, you would have:

```
...
 public:
  LambdaConsumer<int> brightness_{[this](int value) {
   this->set_backlight(value);   
  }};
```

From a class user POV (even if you were the same person 😁), the intent is much clearer if you can do:

```
brightness_producer->connect_to(my_display->brightness_);
```

rather than having two different producers connect to what seems like the same consumer.

Have a look at the [`SystemStatusController`](https://github.com/SignalK/SensESP/blob/main/src/sensesp/controllers/system_status_controller.h) implementation for an example - although the public getters are an unnecessary distraction, IMO. (I can say that because I've written that code. 😁)

mairas, 2024-11-16:
I'm changing this issue into a discussion in hopes that it can be found easier in case someone else is struggling with the same thing.