SignalK/SensESP #802: UnknownPackageError: Could not find the package with 'SignalK/SensESP @ ^3.1.0' requirements for your system 'windows_amd64'

dagnum, 2024-12-22:
Whenever I try to use the latest 3.1.0 on both of my PCs I get this error:
UnknownPackageError: Could not find the package with 'SignalK/SensESP @ ^3.1.0' requirements for your system 'windows_amd64'

Using Seeed S3 Board

[Relay Controller.zip](https://github.com/user-attachments/files/18223488/Relay.Controller.zip)

TMaketa, 2024-12-23:
[Relay.Controller.zip](https://github.com/user-attachments/files/18229142/Relay.Controller.zip)

dagnum, 2024-12-23:
Thank you trying this now...can you tell me what was changed?

mairas, 2024-12-23:
Ahh. Sorry about that! I actually didn't release 3.1.0 yet even though I already anticipated it in the ini file. Change your SensESP dependency from 3.1.0 to 3.0.0 and that should fix it.

dagnum, 2024-12-23:
Ahh thank you! Trying it now!

closed 2024-12-23 by dagnum