SignalK/SensESP #812: Analog input is scale is wrong

cnjboards, 2025-03-15:
I upgraded my existing application from V3.0 to V3.1. It is an engine monitor application.

I am using a built in analog input (36) to scale a pressure reading. 

The code snippet is below: 
(note: OIL_PRESSURE=36)

// Oil Pressure Sender Config //
// transducer has .5V - 4.5V for 0-100psi (689476 Pa).
// Analog input is 3.3V raw with a resister divider of 7.5K and 11K.
// analog input returns 0raw=0V and 4095raw=3.3V @ the pin.
// thru resister divider, .5V xducer ~= .2V at chip = .2*4095/3.3 ~= 248 (125 actual)
// 4.5V xducer ~= 1.82V at chip = 1.82*4095/3.3 ~= 2258
const char *kOilPressureADCConfigPath = "/propulsion/Engine Oil Pressure/ADC/scale";
const char *kOilPressureLinearConfigPath = "/propulsion/oil pressure/calibrate";
const char *kOilPressureSKPath = "/propulsion/oil pressure/sk_path";
const float OilPmultiplier = 228;
const float OilPoffset = -82080;

// Oil Pressure ESP32 Analog Input
auto analog_input = new AnalogInput(OIL_PRESSURE, 500, kOilPressureADCConfigPath, 4096.0);

auto analog_input_linear = new Linear(OilPmultiplier, OilPoffset, kOilPressureLinearConfigPath);
ConfigItem(analog_input_linear)
      ->set_title("Engine Oil Pressure Scale");

analog_input
      // scale using linear transform
      ->connect_to(analog_input_linear)
      // send to SK, value is in Pa
      ->connect_to(new SKOutputFloat("/propulsion/oil pressure", kOilPressureSKPath,
        new SKMetadata("Kpa",                     // Define output units
          "Oil Pressure Kpa")))  // Value description

      // for N2K use, value is in Pa
      ->connect_to(
        new LambdaConsumer<float>([](float engOilPresValue) {
          OilPres = engOilPresValue;
          // if negative then make zero
          if (OilPres <= 0.0) OilPres = 0; // for some reason n2k does not like negative oil pressure??
        }));

Below is the verbose debug output from serial for this sensor:
signalk_delta_queue.cpp: delta: {"updates":[{"source":{"label":"EM V3 Lite"},"values":[{"path":"/sensors/digital in2","value":true},{"path":"/propulsion/engine RPM","value":1773.222},{"path":"/propulsion/oil pressure","value":1.960344e8}]}]}

Notice the output value for oil pressure is now 1.960344e8 which very wrong.
Everything else is working as expected.

cnjboards, 2025-03-15:
This is what the debug log looks like in V3.0, no code change, same signal inputs.
signalk_delta_queue.cpp: delta: {"updates":[{"source":{"label":"EM V3 Lite"},"values":[{"path":"/sensors/digital in2","value":true},{"path":"/propulsion/engine RPM","value":1770.319},{"path":"/propulsion/oil pressure","value":181672.6}]}]}

hurskmar, 2026-01-09:
I have tried to implement a tank sensor by using the AnalogInput and I have similar findings - when I directly read values from the analog pin (ESP32C3), it returns a logical and correct value, but not by using AnalogInput.

hurskmar, 2026-01-11:
The AnalogInput-class is now deprecated. I started to use RepeatSensor instead and it works fine: 
auto* analog_mvolts = new RepeatSensor<int>(
      kReadInterval, 
      []() { return analogReadMilliVolts(kAnalogPin); }
  );

closed 2026-09-11 by mairas